BeFlE/SoMoSe #2: Meaning of referenceWet and referenceDry values

Companion83, 2022-04-21:
Hi,

I don`t understand the meaning of the reference values for dry and wet. When testing the your sensor I get values between 0 (dry condition) and 120 (pure water), moist soil lies between these.

Best regards

BeFlE, 2022-04-21:
Hi,

the reference values normalize the measured signal from the sensor to get a humidity value of 8 bits.
The frequency coming from the sensor is measured, better said the period duration.  

In detail 16 periods are measured. The time that passes gives information about the humidity, but must be further processed to obtain a meaningful value.

The sensitivity and the measuring range of the sensor can be influenced with the reference values.

Humidity(8 Bit) = (((period duration) - reference dry) / reference wet) * 255

A new data sheet will be published soon, which will provide more information on this topic.

Many greetings

Companion83, 2022-04-26:
Thank you! That clarifies it a lot!

josefmaierfl, 2023-02-14:
Hi Christian,
I am currently taking a closer look on your sensor to understand, how the measurement is performed. I understand that you measure the frequency which changes according to the change of the capacity. But how did you realize the oscillating circuit? Did you use an Inverter with Schmitt trigger, a variant of the 555 timer IC, or something else?

Thanks,
Josef

BeFlE, 2023-02-14:
Hi Josef,

exactly, a Schmitt-Trigger inverter is used to realize the oscillating circuit.

Many greetings
Christian

josefmaierfl, 2023-02-14:
Thanks a lot.

derjape, 2024-10-20:
but do I have to adjust the wet&dry reference values for every sensor on my own?

what is the procedure for this? 
Use the raw-value of a completely dry sensor as "dry-reference"
And the raw-value of a wet sensor (e.g. within a glas of water)  as "wet-reference"