BeFlE/SoMoSe #3: small bubbles on the sensor

Nicky47, 2022-06-14:
Hey on my sensor bought on amazon a few weeks ago are apearing small round hard bubles which you can feel with your fingers. Which kind of insulation material is used ? could it be possible that small microorganisms like bakterial cultures from the soil could diggest the epoxi or whatever is used ? 

Also i read the value over the analog signal and get a reference value of around 1024 if put in pure water and around 9 dry on air. When i water the plants the value will be around 300 after watering and then increase till around 900 when its time to water again. Could the microorganims in the soil affect the capacitiv value because they have a conduction value ? 

my tomates dont get any mineralic fertilizer instead i work with soil bacterias and compost soil. 

if needed i could send pictures of the small bubbles. Did anyone had similar problems ?

BeFlE, 2022-06-14:
Hi,

thank you very much for your message.
An epoxy resin is used as the protective layer.
We are not aware of a similar case where these bubbles occur, we will try to find the cause.
Is the epoxy resin coming off or dissolving?

Do I understand correctly that if you do not have the sensor in the ground but only in pure water or in the air, the measurement works correctly, but in the ground a wrong behavior occurs?
That would be a very strange behavior and is not yet known to us.

Can you send us some more information about your setup so that we can try to reproduce these problems?

I look forward to your reply soon.

Many greetings from Dresden
Christian

Nicky47, 2022-06-20:
Hey Christian,  vielen dank für die kleine Code Demo. Ich frage mich nur noch wie ich „SoMoSe_Arduino.h" Bibliothek installiere ? Ich bekomme immer die Fehlermeldung „SoMoSe_Arduino.h: No such file or Directory“ im Bibliotheksmanager hab ich Sie leider nicht gefunden. Wäre super wenn du mir hier vllt auf die Sprünge helfen könntest.  Alles klar super wäre fantastisch wenn du mir kurz bescheid geben könntest wenn die Sensoren wieder auf Lager sind.  Viele Grüße  Nik  Gesendet von Mail für Windows Von: BeFlEGesendet: Mittwoch, 15 Juni 2022 15:07An: BeFlE/SoMoSeCc: Nicky47; AuthorBetreff: Re: [BeFlE/SoMoSe] small bubbles on the sensor (Issue #3) Hi Nik,ich habe auch die Vermutung, dass der Sensor durch die Organismen beeinflusst wird und das Epoxidharz dadurch angegriffen wird.Aktuell ist uns kein Fall bekannt, bei dem ein ähnliches Verhalten auftrat.Wir werden aber versuchen die Ursache zu finden und ein Lösung zu erarbeiten.Das Beschrieben Verhalten bei der Feuchtemessung ist sehr merkwürdig. Offensichtlich funktioniert dein Sensor ja ordnungsgemäß, das Verhalten in der Erde muss also andere Ursachen haben.Der Messwert der über die I2C- Schnittstelle ausgegeben wird ist identisch mit dem Wert auf der analogen Schnittstelle.Eine Änderung des Verhaltens ist also nicht zu erwarten.Ich habe trotzdem eine kleine Demo für den Arduino geschrieben und hier: https://github.com/BeFlE/SoMoSe/tree/main/Arduinoveröffentlicht. Leider konnte ich mangels Hardware den Code nicht testen, ich hoffe es hilft dir trotzdem weiter.Im Moment ist unser Lager komplett leer, eine neue Charge ist aber in Produktion und ist bald auf Amazon verfügbar.Ich freue mich bald wieder von dir zu hören.Viele GrüßeChristian
	
      
		Virenfrei. www.avast.com

BeFlE, 2022-06-20:
Hi Nik,

mir fehlt leider auch die Erfahrung mit der Arduino IDE, ich nutze normalerweise komplett andere Dinge.
Habe die Demo nochmal angepasst, so sollte es auch funktionieren, ohne dass die Bibliothek eingebunden sein muss.
Gern gebe ich dir bescheid, wenn die Sensoren wieder verfügbar sind.

Viele Grüße
Christian

BeFlE, 2022-06-23:
Hi,

es sind neue Sensoren auf Amazon verfügbar.

Viele Grüße
Christian

Nicky47, 2022-10-11:
Hey Christian,

Vielen lieben Dank für das ändern des Codes. Darf ich fragen mit was du
sonst arbeitest? In der Uni haben wir gerade begonnen mit Labview und
Matlab zu arbeiten. Mit der Arduino ide spiele ich nur Zuhause etwas herum.
Super dann freu ich mich auf deine Nachricht.

Viele Grüße

Nik

BeFlE, 2022-10-14:
Hi Nik,

ich arbeite hauptsächlich mit PIC- Mikrocontrollern der Firma Microchip und der dazugehörigen IDE MPLAB X.
Der Einstieg ist dort zwar etwas schwieriger, m.M.n. hat man aber bessere Kontrolle über die Hardware und eine sehr große Auswahl an Controllern und Zubehör.

Viele Grüße
Christian

cameo69, 2022-11-02:
Hi, I have a sensor showing similar bubbles.
[image]
The one in the back never touched soil, it was not used yet, while the other with the bubbles was stuck in earth to monitor my tomatoes.
Both sensors were bought in July 2022 from Amazon (de).
Here is another shot
[image]

BeFlE, 2022-11-02:
Hi,

thank you very much for your message.

I am already in contact with the manufacturer of the epoxy resin.

In my own tests I was able to reproduce these bubbles. However, they do not seem to spread further and the resin does not dissolve.
So there is no reason to worry for the time being.

Many greetings
Christian