BeFlE/SoMoSe #9: Using analog out with an analog digital converter

PhillipPapastefanou, 2023-06-06:
Hey, 
I have set up your sensor with my raspberrypi via I2C. However, the signal that I get is a little bit noisy. Is it possible to use the analog output of the sensor with an ADC such as MCB3008?
Cheers
Phillip

BeFlE, 2023-06-07:
Hi Phillip,

yes it is possible to read out the sensor also analog. The only prerequisite is that you have purchased a sensor with a pin header. In the version with cable (SoMoSe v2.2w) the analog pin is not led to the outside.
Can you tell me what kind of interference you have on the digital lines and what causes this?

Many greetings
Christian

PhillipPapastefanou, 2023-06-07:
Dear Christian, 
thanks for your super quick reply!
I have both sensors the one with and without the cable. I will then connect to this analog output pin to my ADC and then see how that works. What's put out via the analog output? Just the raw signal of the soil moisture sensor? or also the soil temperature? Sorry, I am quite new to all these sensors and output signals... 
Regarding the interference: So I was looking at the raw soil moisture data from the sensor via I2C and it appeared to me that there were quite some oscillations of the sensor data. I can post a figure in case you want to see it.
Would you recommend running the sensor with 5V instead of 3.3 and using a logic level converter for the output signal? Or does that not really make any difference?
Thank you again very much for your help!
Cheers
Phillip

cameo69, 2023-06-07:
May I ask what type of cable you used for I2C and how long it was? I think the problem with I2C is that it is not designed for longer cables. In my experiments I had everything longer than 50cm fail and even with 40cm it was unstable, until I separated SDA and SCL with GND in between.
What I would like to try out is, if it can be mitigated with a differential signal buffer like the PCA9615DPZ. But I didn't find the time so far.

PhillipPapastefanou, 2023-06-08:
I use it with the normal dupont cables which are about 50cm long... That might explain the noisiness. Thanks for sharing the idea with a signal buffer. Do you also think that something like a LTC4311 I2C Extender / Active Terminator could improve?
I have quite some time a hand and could some of the tests, hehe :) 
Thanks for your help again!
Just once again my question what exactly does the analog output return? soil moisture or temperature? or both?

BeFlE, 2023-06-08:
Hi, 
The humidity value averaged over 30s is output at the analog output.
Everything is also in the data sheet:
[https://github.com/BeFlE/SoMoSe/blob/main/Docs/Datenblatt_SoMoSe_v2.2_neu_1_EN.jpg](url)

But once again to the I2C bus. 
I have already connected I2C sensors over 10m standard 4-wire cable without any problems, at approx. 70KHz clock.

If you have strong interferences on the signal, this has possibly other causes. 
Which pull-up resistors do you use?
How fast is the I2C clock?

Many greetings
Christian

cameo69, 2023-06-08:
> Do you also think that something like a LTC4311 I2C Extender / Active Terminator could improve?

I read this https://www.adafruit.com/product/4756 and it looks fairly promising. There is even one large advantage, you would not need to modify with the far end. The PCA9615DPZ works by putting the mirrored signal on another line to cancel out interferences at the end, much like an XLR cable in audio technology.
If you have the time I would give the LTC4311 a shot first.

What Christian said is also interesting. I don't know what I2C speed I used, but it also sounds worth a try reducing the speed.

cameo69, 2023-06-08:
> https://github.com/BeFlE/SoMoSe/blob/main/Docs/Datenblatt_SoMoSe_v2.2_neu_1_EN.jpg

Link not working for me; [this](https://github.com/BeFlE/SoMoSe/blob/bb0ea71dacf349cce503de53a21221859aaf67ba/Docs/Datenblatt_SoMoSe_v2.2_neu_1_EN.jpg) should work.

PhillipPapastefanou, 2023-06-09:
Thanks guys for your quick comments and help!
I will play around with the setups and let you know how it works!
Cheers
Phillip