BeFlE/SoMoSe #10: Initialization time of ~600ms

andyfr, 2023-07-01:
Hi,
I want to build a battery powered moisture sensor with an ESP32.
As it is battery powered I'm putting the ESP32 into deep sleep and power it up once every hour to save energy.
I noticed that the sensor needs roughly 600 ms after powering up, before it measures any useful data. I'm using the I2C protocoI to measure the values. I would like to reduce the time to save more energy.

Are the 600ms by design, or am I doing something wrong?

BeFlE, 2023-07-03:
Hi,

the sensor needs some time until the first measured value is available.
On the one hand, this is because the (slow) CPU has to boot and on the other hand, the oscillating circuit has to stabilize for the measurement. Then the proper measurement has to be done and calculated.

So you don't do anything wrong.

An low-power version, for exactly such purposes, I have already in planning.

Many greetings
Christian

FHEMbeta, 2024-04-22:
> Hi,
>
> the sensor needs some time until the first measured value is available. On the one hand, this is because the (slow) CPU has to boot and on the other hand, the oscillating circuit has to [...]

I have bought two of the SoMoSe BeFIE sensors and running them with a solar powered battery on ESP8266. In comparison to a Trübner SMT50 sensor I noticed a really high (around factor 20) power usage of the SoMoSe BeFIE sensor even when the ESP is in deepsleep mode.

Waking up the ESP all 30 minutes to publish the current readings via MQTT continously discharges the 4000mAh Li-Ion battery which is charged by a 0.6W solar panel.

So I decided to power off the sensor in deep sleep with a transistor to get the current draw down. This works ok for some days, but needs more extra logic, wiring and I don't know if it has some drawbacks over time. One point is that after cutting off the voltage and turning it on again, the SoMoSe BeFIE sensor needs about 30-40 seconds to get stable values via 3.3V. Otherwise the results are really unusable with moisture over 100% (while it is absolutely dry - 3% after 40 seconds).

What's the current state of the low-power version? I need a lot of sensor for my outdoor plants so that I can connect them to the automatic watering system. Would like to test if the low power sensor works without modification with battery and solar powered. Would like to connect 5-10 sensors to one ESP8266 with different addresses.

FHEMbeta, 2024-04-22:
Thanks for your quick reply.

What's the current consumption of the new sensor? Is it possible to get a sample (country DE) for testing purposes?

FHEMbeta, 2024-04-22:
@Elektroarzt Can you please upload the STL files? Would bei great.

Elektroarzt, 2024-04-22:
I thought about publishing it. At the moment it's beta. Give me some time to test the solution indoors and outdoors and check compatibility with the new SoMoSe (V3.0?). Will be with step-down 5...40V and epoxy sealed  electronics. Here a little teaser:

[image]

FHEMbeta, 2024-04-22:
Please publish, it looks really nice.

I soldered the wires and finished with some fluent plastic (called Microdip). But ist's much more work than printing a 3D file and doesn't Look as nice as your solution.

You could declare the file as Beta and update it on ThingIverse as soon as progress continues.

BeFlE, 2024-04-22:
Hi,

the PCB of the new version is 99% compatible with version 2.2, only the upper holes are different. I have just uploaded the new .stp file.
Otherwise, the new version is completely backwards compatible with the old version.

About the power consumption:
In continuous mode, the sensor requires approx. 2.2mA.
In low power mode it requires approx. 800uA (standby) and 1mA (measurement). I am still working on this at the moment.

For comparison, the v2.2 requires approx. 25mA.

For samples please write me an e-mail to info@befle.de, I think I can spare a prototype ;)

Best regards
Christian

Elektroarzt, 2024-04-22:
@BeFlE: Thanks for the information, that's good news :grinning:
@FHEMbeta: Concerning the housing, it is tailored to some components I use for mains power supply (e.g. ESP32-C3 Supermini, buck converter, etc.). So it will not fit for your Devboard nor a battery pretty sure. Are you fixed on a special board? If not, we could do some development on teamwork for a SoMoSe companion project. I would fork the project and you could bring in your insights of a battery version. I could bring in my mains power supply version including the 3D design of a housing. What do you think?
I'm also tinkering on a water valve system with flow meters for irrigation.

Sorry for spamming this issue off topic, we should move this conversation asap.

BeFlE, 2024-04-23:
Just a little update:
I have now preferred to deal with the LowPower mode again. The standby consumption could be reduced to 16uA.

BeFlE, 2024-05-06:
Production is expected to be completed by the end of calendar week 20. Orders can then be placed by e-mail. It is expected to be available on Amazon from the end of May.

Elektroarzt, 2024-05-08:
Will your ESP32 firmware support deep sleep to have the whole system energy saving beside the 16uA of the sensor?

BeFlE, 2024-05-08:
Currently, deep sleep is not yet supported.
However, it is in progress and will be done next.

Elektroarzt, 2024-05-15:
@FHEMbeta: the whole project including housing and a system PCB maintaining an ESP32 is released under [this Link](https://github.com/Elektroarzt/floragrow)

FHEMbeta, 2024-05-21:
@Elektroarzt Really great project! I am really looking forward to the battery powered version (with solar?), as I have a lot of plants without a power connection next to it.

Elektroarzt, 2024-06-04:
I think solar powering is not suitable for ESP32 in Central European regions with small solar panels that fit to the sensor size. I am tinkering on a battery version with Qi charging possibility to keep the water resistance.

Elektroarzt, 2024-06-04:
That sounds great, really looking forward to it.

FHEMbeta, 2024-06-04:
I think that should work with a small solar cell powered, too.

I am running multiple ES8266 boards waking up every 30 minutes, measuring, publish via MQTT and deep sleep. They are continously 100% charged.

I tested an ESP32-C6 (very low consumption in deep sleep) with the new v3.4 sensor. But I am really getting non consistent results, no results from the sensor or it stops working. Maybe the ESP32-C6 is currently not really stable supported. Debugging via Arduino serial monitor does not work (sadly Windows 11). Reading firmware and hardware version is very unreliable.

@BeFlE Could you please publish a PCB + housing with a 18650 or 21700 Li-Ion cell and mini solar cell? That should run forever.

BeFlE, 2024-06-04:
@FHEMbeta That is the intention. However, I have decided on the LIR2450 Li-Ion cell for the time being, as I want to save as much space as possible. I think this cell should provide enough of a buffer to bridge longer periods of unsunny weather. But only the tests will show that when the prototypes arrive.

FHEMbeta, 2024-07-30:
@BeFlE Any updates on your project with the Li-Ion cell? I would like to buy multiple sensors and put each in a small case with battery (+solar) that run independently, My current setup with IP65 cases is not really nice, too big and impractical. Placing the sensors with a small footprint would be really nice (for example on the balcony in small boxes, vegetable patch, bananas and palm trees).

BeFlE, 2024-07-30:
Hi,

The project is going quite well, I have been testing the hardware for about 2 months and it has worked without any problems so far.
Unfortunately, I don't have enough time at the moment to get the whole thing ready for production.
The PCB is about 25x45mm in size, the built-in LiIon cell can easily operate several sensors even under bad light conditions.
If there is general interest, I can calculate the costs for a small series and have some PCBs produced.
Best regards Christian

FHEMbeta, 2024-07-31:
Do you have a solar panel, too? Which size does it have?

I am really interested in a small footprint design running without to replace the battery. 3D printing is no problem. A clean PCB without wiring on small room would be really nice.

Do you have some pictures and a little bit more information about this?

Elektroarzt, 2024-07-31:
Interested too 😁
I could help in making a compact system from it in Fusion 360 for an easy DIY solution.

Elektroarzt, 2024-07-31:
@BeFlE if you could provide a sample, I could develop the companion parts in parallel? Just a thought

Elektroarzt, 2024-07-31:
No, I built an experimental model on basis of a Li-Ion cell and Qi wireless charging. I plan to use the sensor indoors and I'm not so confident if a solar version will work for me in this case.

FHEMbeta, 2024-07-31:
@BeFlE Do you know any other retailer to buy this solar panel (is really smaller than the smalles I know from AliExpress)? Estimated delivery time is 13 weeks and have to buy 13 pieces to get a "good" price.

BeFlE, 2024-07-31:
@FHEMbeta No, unfortunately I don't know of any distributor who has such or similar cells available. 
At the moment, however, there are enough in stock and ready for immediate shipment, the delivery time of 13 weeks refers to the manufacturer's subsequent delivery.
I know about the annoyance with the high shipping costs, but unfortunately there are very few alternatives in Europe.

FHEMbeta, 2024-07-31:
You are right, more than enough in stock.

Could you please post all hardware which is necessary for your prototype. I would build 5-10 sensors depending on the total cost of each one.

Do you have STL files for the case fitting with the SM101K07L?

BeFlE, 2024-07-31:
You only need my PCB, a LIR2450 Li-Ion cell and a solar cell, and possibly a housing.

Unfortunately I have no further data on the solar cell, only what can be found online. Maybe on the manufacturer's website?

BeFlE, 2024-07-31:
As promised, here are the pictures...

[image]

[image]

[image]

BeFlE, 2024-08-01:
@Elektroarzt  I have uploaded the [3D model of the PCB](https://github.com/BeFlE/SoMoSe/tree/main/3D). Please note that it is not yet the final version and things may still change.

FHEMbeta, 2024-09-19:
Are there any news on the solar powered version? Do you have a STL file to print?

I would like to convert my big housings with 21700 batteries to a small footprint one with one ESP per plant to monitor and no cables between. At the moment I am running 3 of v3.4 SoMoSe and planning to buy a lot more as soon as the all in one solution is available and working.

Elektroarzt, 2024-09-19:
I just played around a bit with it, but made no design from it. As @BeFlE stated, there is no final version up to now. So if I would make a shot, the PCBA you would get from him would most likely not fit.
@BeFlE I found no mounting holes or something similar, is that intended? For sure I could design something like clips or so, but to be on the safe side, maybe two M2.5 holes could help. 3D printing and clips are kind of enemies when printed in Z-axis direction ;). Half or 1/3 holes on the edge could also be a good compromise for mounting and PCB space / cost.

BeFlE, 2024-09-19:
I have decided to change the whole thing a bit. I have enlarged the whole board a bit, now it is not only usable for the SoMoSe, but also as a normal ESP-Dev board for other projects. 
An ESP32-C3-Mini is used, all pins are routed to pin headers, which can be soldered on if required. There is also a USB-C port to operate the board and charge the battery if necessary. Some pads of the pin header have been enlarged and positioned so that a SoMoSe can be soldered on directly. All available voltages (5V_USB, 3V, solar cell voltage and battery voltage) are routed to the pin headers.
Solar cell voltage and battery voltage are divided down previously by voltage dividers and can be measured by the ESP if required.
A solar cell can be plugged in directly to create a compact board (this requires an additional PCBA).
There is a battery holder on the back for a LIR2450 cell, but another cell can also be attached via the pin header if required.
In the next few days I can provide a few pictures and possibly 3D files in advance (not final).
About the schedule:
The development will soon be completed.
After that I will have a (final) prototype made.
If everything works as desired, a series can be produced.
I expect to start testing the prototype in mid to late October.
A first series will therefore be available in November at the earliest.
@Elektroarzt Thanks for the hint about the drill holes, I have taken this into account.

I look forward to your feedback.
Best regards Christian

BeFlE, 2024-09-20:
Here are the pictures, I have stored the 3D files in the 3D folder.

What do you think? any more suggestions?

[image]
[image]
[image]
[image]

Elektroarzt, 2024-09-20:
Unfortunately I can't open Solidworks files, so I just had a look on the pictures. In the following I write down what comes in my mind, when I see the board as a general purpose dev-board. I don't know if it matches your origin idea, space, routing or cost constraints. So please just take it as ideas, that maybe make no sense from the inventor view.

1. I would place the contacts for the SoMoSe Board so, that the whole system will result more streamlined. If I connect the current board and SoMoSe in mind, the system will result in a very big unsymmetrical housing, probably not suitable for all flowerpots. Maybe 90° rotation of the SoMoSe contacts would be a good position, so the overall system is more high than broad. This would give the chance for the solar panel to reach more out of the flowerpot and get more light. The other free GPIOs could stay at the side, as placed in the current design.

Current system:
[image]

Proposal:
[image]

I know this would unfortunately destroy big parts of your current design 🙈.

3. If possible, a SMA connector for an optional external antenna would be great to enhance WiFi range, if needed. The D1 mini Pro had a 0R resistor to choose between internal and external antenna for your reference.
[image]
4. The switches could be rotated by 90° to save PCB space, what I see from the pictures.
5. An onboard WLED would be super nice and low cost.
6. Having a general purpose switch is a feature I always add to my designs. I use the same as you already designed in for reset and boot.
7. Some components have no reference prints, maybe this is from beta state.
8. If there are more unused GPIOs of the ESP32 mini, I would route them out on the edge.
9. Additional half THT pads on the edges would give the possibility for optional SMD soldering like the XIAO ESP32-C3 has. I love this feature, as it results in super compact systems and is by the way dirt cheap.
[image]

All in all I love your design and the features it brings out of the box ... really well done, respect! 🤩
What LCD do you plan? If you could upload STEP files, I could take a more detailed look on the design.

Elektroarzt, 2024-09-20:
An additional idea, that is a point when it comes to IP68 usage outdoors:
I tinkered around with a QI charging board, that is able to deliver 5V / 600mA for charging the battery. It could directly replace the 5V bus voltage of the USB-C port. To achieve a water resistant housing, a QI charging circuit would be the ideal way in my opinion, as every hole in the housing makes potting a rocket science unfortunately. I made really a lot of tests with resin, different versions of 3d printed housings and different ways of sealing against leakage for the potting process. Best is to have just one opening for the sensor, that is positioned at the top while potting. I successfully tested this board with an ESP32:
[image]

https://de.aliexpress.com/item/4000086798457.html?spm=a2g0o.order_list.order_list_main.4.3c7b5c5fewfiEc&gatewayAdapt=glo2deu

BeFlE, 2024-09-23:
@Elektroarzt Thank you for the helpful feedback.
I have had a few thoughts on your points:

1. that was also my first intention, but I didn't want to destroy the integrity of the PCB antenna. I now have a different approach, see pictures.

2. did I overlook it ? ;)

3. the used ESP32-C3-MINI has a drop-in alternative with u.fl-connector instead of PCB-antenna, so this can easily be produced in 2 variants.
I don't think the approach with the 0R is so good, firstly because the integrity of the RF path is not guaranteed, and secondly because I would have to go to a lot more trouble because I can't use a module.
4. the buttons are almost square, the 3D model is just not quite correct.
5. yes, a WLED would be great. I've looked into it, but unfortunately it's only practicable with a lot of effort. The main problem is that I only have one defined voltage available (3V) and such an LED requires 3.5V or more. 
6. unfortunately there is no more space for an additional switch, but this can easily be added by the user if desired.
7. I still have to finalize the silcscreen
8. all possible GPIOs are already routed to the pin headers
9. the idea with the pads actually seems quite good. however, you have to remember that the battery holder is on the bottom and the PCB can therefore not be soldered as an SMD to another PCB.

I have not planned an LCD, that would also go beyond the scope. The whole board is mainly intended for wireless sensor applications. Of course, other things with low energy consumption can also be realized.

The idea with the Qi charger is good, but I think it's too much. In principle, the solar cell does nothing else. However, the user could also add this functionality later. 

Unfortunately, I can't generate STP files as nice as the .wrl, but I can make them available to you. But then you'll only have the essentials.

[image]
[image]

Elektroarzt, 2024-09-23:
Thanks for your information, some answers below.
1. Understood, this seems a good compromise ... by the way, do you maybe have a pinout issue? My sensor has the order AOUT, GND, 3V3, SDA, SCL vs. your design with GND, 3V, IO4, IO5, IO6 on the silkscreen
2. You didn't overlook it, I missed it somehow 😉
3. Good
4. Good
5. Maybe just a single color LED could fit into the design, if you like the idea
6. Good
7. The blue box with the two connectors on the top layer looked like a display, then I don't know what this is
8. Good
9. If I would use the board as a Dev-Board, I would cut out my "motherboard" for the battery to solder down the Dev-Board via the SMD pads.
10. QI: I thought about that after sending the message out. Integrating this is no problem for me, as it just connects to the 5V, if I like to.

Elektroarzt, 2024-09-23:
By the way ... will the board be open or closed source? Looks like KiCAD

BeFlE, 2024-09-23:
1. very attentive ;-) was indeed wrong.
5. I'll see if I can find a little place for it
7. the blue box with the plugs should be a plug-in solar cell module
9. i didn't think of that possibility. i'll try to adapt it.

I haven't thought about whether it should be open or closed source.

Elektroarzt, 2024-09-23:
You have my vote for open source 😊

Elektroarzt, 2024-09-23:
One last point ... more a question of the schematic design. Is your converter from solar power to 3V3 a buck type or buck / boost? This could enable for much longer battery times. I'm thinking of a TPS63020 for example.

BeFlE, 2024-09-23:
It's just a buck. But at 3V instead of 3.3V. At 3V the battery is already pretty empty and normally the solar cell should keep the battery at a higher level. I have already tried this in a test and it actually works quite satisfactorily. I chose this mainly because this buck converter has an extremely low self-consumption. it cannot be compared with a buck/boost.

BeFlE, 2024-12-09:
Hi,

I have an update for anyone who is still interested.
Development is now complete and I'm in the final preparations for the pilot series. 
This is expected to be ready by the end of January.
A corresponding data sheet will be published in the next few weeks. 

Here are a few key data:
- Can operate standalone via USB, just like any other DevBoard
- Internal battery (LIR2450) or external LI-Ion battery can be attached
- Solar cell can also either be plugged on directly (PCBA from BeFlE) or an external cell can be attached
- Current consumption in SLEEP approx. 40µA
- LED for free use
- Switchable 3V output for supplying peripherals
- footprint provided for a reed contact, so that ESP can be woken up magnetically without opening the housing

[image]
[image]

FHEMbeta, 2024-12-12:
Great to see that you continue to develop the board. I am really looking forward to use the all in one sensor for my plants without having a cable connected to an IP box.

BeFlE, 2025-02-19:
It has taken a long time, but now the time has come. The circuit boards for the ESP board have finally arrived. A corresponding repo can be found [here](https://github.com/BeFlE/ESP32-C3-DevKit-EH).
The missing data will be completed in the near future and it will soon be online on Amazon.
If there is already a need, please mail me, info@befle.de

Best regards
Christian