BeFlE/SoMoSe #11: 3D Step Model of SoMoSe

Elektroarzt, 2023-07-06:
Hi SoMoSe team ... I'm implementing your great sensors in an automatic plant watering system for several plants and like to develop a 3D printed housing for the electronics. Assuming you have all 3D data, could you upload a STEP file or something similar of the whole device for me to construct a housing around this model? In return I could publish the housing containing SoMoSe, ESP32 D1 mini and a step-down converter.
Thanks in advance!

BeFlE, 2023-07-06:
Hi,

I have uploaded the STEP file in the "3D" folder.
If there are problems with it, I look forward to a feedback.

Many greetings
Christian

Elektroarzt, 2023-07-06:
Looks good, thanks for the fast response!

closed 2023-07-06 by Elektroarzt