BeFlE/SoMoSe #14: ESP-IDF Integration

goronlant, 2023-10-31:
Hallo zusammen,

seit 2 Monaten beschäftige ich mich mit dem Sensor und wollte jetzt noch einen Schritt weitergehen und das alles in der ESP-IDF-Umgebung zu programmieren. Jedoch fehlt es mir die Bibliothek/eine Einführung, wie man das Ganze mit dem ESP-IDF verbindet. 
Könnte mir jemand mal helfen? Auf eine Antwort freue ich mich.

Grüße

##############

Hello,

For 2 months, I've been working with the sensor and now I want to take it a step further and program everything within the ESP-IDF environment. However, I'm missing the library or an introduction on how to integrate this with the ESP-IDF. Could someone please help me with this? I'm looking forward to a response.

Best regards.

goronlant, 2023-11-03:
I've done it myself. Thanks anyway!

closed 2023-11-03 by goronlant

allesgutewarweg, 2025-01-20:
Hallo,

ich versuche seit einigen Tagen ebenfalls das ESP-IDF in ESPhome zu verwenden da dieses wesentlich runder mit meinem Arduino Nano ESP32 zusammenarbeitet.

Wäre es möglich deine konvertierte Bibliothek zu erhalten? Ich komme leider nicht alleine zum Ziel.

@BeFlE sollte sich @berkayozg melden wäre es möglich das ihr bei der Sache unterstützt?

Viele Grüße und Danke für diesen tollen Sensor!

BeFlE, 2025-01-20:
Hi,
I have placed a code for ESP-IDF here:[ESP-IDF](https://github.com/BeFlE/SoMoSe/tree/995a6c07d71d9373ece058aaed68cd39b1a6b33c/ESP-IDF)

The whole thing is AI-converted and unfortunately I didn't have a chance to test it. 

best regards 
Christian

allesgutewarweg, 2025-01-21:
Hello Christian,

thank you very much!

I had tried to do the conversion myself with ChatGPT, unfortunately without success. I will test your code shortly and give you feedback here. This will probably be over the weekend.

Best regards,
Maik