BeFlE/SoMoSe #16: bin file for ESP32-C3

Elektroarzt, 2024-04-14:
Hi Christian ... I got several sensors and like to run your compiled firmware, which is made for the ESP32. To have a nice compact design, I like to use a very tiny board like XIAO ESP32-C3 or the ESP32-C3 super mini from Ali. Could you compile and provide a C3 version therefor?
Many thanks in advance!

BeFlE, 2024-04-15:
Hi Elektroarzt,

Thank you for your message.
I will have a look at this later in the week and try to do so. 
I'll get back to you as soon as possible.

Best regards
Christian

BeFlE, 2024-04-18:
Hi Elektroarzt,

I have adapted and compiled the firmware for the ESP32-C3. 
I tested it with a "Waveshark ESP32-C3-Zero", everything works great.

I look forward to your feedback.

Best regards
Christian

Elektroarzt, 2024-04-20:
Works like a charm on ESP32-C3 Super Mini 🎉 ... congratulations to this really useful piece of software!
As a small improvement, home and update section could show the current version installed and the update log could show the available version on the update server to have a choice before updating based on the changelog.

closed 2024-04-20 by Elektroarzt