BeFlE/SoMoSe #17: Multiple sensors on esp32 with HomeAssistant

opiswahn, 2024-05-08:
Hi, 

maybe I just don't understand it, maybe the documentation is lagging somewhat, so please help me  :D
I'm trying to connect 3 sensors to an ESP32 over I2C on Pin 21 and 22. No additional pullups. I'm using esphome (HomeAssistant) on the esp32. If I only activate 1 sensor i n the configuration et seems to work (humidity values seem to be a bit off not that's maybe another topic). If I try to activate another one (with a different I2C address), I don't get any values from it (nor additional ones)

Here's the relevant part from my HomeAssitant  yaml:
[image]

So what am I missing here?

Greets from Gernany!

sk1kn1ght, 2024-06-07:
Take what i say with a grain of salt cause like you i am also a beginner at this. 

What you need to do is program the second sensor to the address that you want. They come (Christian can correct me if am wrong) pre-programmed at address 0x55, and what "auto somose2 = new etc..." do is simply tell HASS where to look for the info. it doesnt actually register it as 0x56.

After a lot of trial and error (and some tears mind you) trying to do it via home assistant i gave up and since i had another micro controller (both arduino and esp will work on this (arduino has specific pins for i2c communication) ) i booted up an arduino ran an i2c scanner since i didnt know which address it had(reason was cause trying to change the address with HASS led it to registering it to 0x2A), and then reprogrammed the address to 0x56.

below samples of code for scanner and i2c reprogrammer. Keep in mind the reprogrammer might not work. when i saved the file, it got corrupted so i had to re-write it but hadnt tested it, but the general idea is there.

**I2C scanner**

[https://learn.adafruit.com/scanning-i2c-addresses/arduino](url)

**I2C reprogrammer**

`// SPDX-FileCopyrightText: 2023 Carter Nelson for Adafruit Industries
//
// SPDX-License-Identifier: MIT
// --------------------------------------
// i2c_scanner
//
// Modified from https://playground.arduino.cc/Main/I2cScanner/
// --------------------------------------

#include <Wire.h>

// Set I2C bus to use: Wire, Wire1, etc.
#define WIRE Wire

void setup() {
  WIRE.begin();

  Serial.begin(9600);
  while (!Serial)
     delay(10);
  Serial.println("\nI2C Scanner");
}

void loop() {

Wire.beginTransmission( 0x60);
int error = Wire.endTransmission();
if( error != 0)
{
  Serial.println( "The sensor is not at 0x60");
}
else
{
  Serial.println( "The sensor is found, changing I2C address");
  //Wire.beginTransmission( 0x60);
 // Wire.write( 0x53);  // password register
 // Wire.write( 0xAA);  // password
 // Wire.endTransmission();

  delay(10);    // not described somewhere, just for safety

  Wire.beginTransmission( 0x60);
  Wire.write( 0x00);  // I2C address register
  Wire.write( 0x50);  // new I2C address
  Wire.endTransmission();
}
  delay(5000);           // wait 5 seconds for next scan
}`

After that both sensors were recognized in esphome.

@BeFlE , EspHome custom component is deprecated. Can we count on you to update the code to support external component? [https://esphome.io/custom/custom_component.html](url)

Virtualizer, 2024-07-03:
> **I2C scanner**
> 
> [https://learn.adafruit.com/scanning-i2c-addresses/arduino](url)
> 
> **I2C reprogrammer**
> 

The code for reprogramming does not work.
After modifying the part where the new I2C address will be written, it works (adapting the repositories Arduino code):
```
Wire.beginTransmission(0x55);
Wire.write(0x41);  // I2C address register
Wire.write(byte((0x56 << 1) & 0xFE)); // new address, 7-bit
Wire.endTransmission();
```

Have 4 working sensors on one ESP32 now.