BeFlE/SoMoSe #19: Standby Mode v3.4

user20240627, 2024-06-27:
Hi,

in the datasheet for hardware v3.4 is specified a standby current of ~16uA.

My question is how to enter this state? Command 0x4C drops the current only to ~800uA.

BeFlE, 2024-06-28:
Hi,

the standby current of 16 uA can only be reached with the latest firmware version (SoMoSe v3- FW version 1.4).

The currently sold lot is still with version 1.3.

An update must therefore be performed.
You can do this with the help of an ESP32 and my [SoMoSe firmware](https://github.com/BeFlE/SoMoSe/tree/main/SoMoSe-MQTT-ESP32_Firmware) or send me the sensors and I will do it for you.

Best regards
Christian

poker-spt, 2024-07-22:
@BeFlE Could you please provide the firmware version 1.4 for the ESP32-C6 hardware?

Which ESP32 did you test working fine?

BeFlE, 2024-07-23:
Hi,

unfortunately I am not able to compile the firmware for the ESP32-C6.
However, it has been tested and provided for the ESP32 and ESP32-C3.

poker-spt, 2024-07-26:
@BeFlE Could you please provide the full code for the firmware. Then I could build it for my newer hardware.

BeFlE, 2024-07-28:
@poker-spt please write me an [E-Mail](info@befle.de) about this topic.