BeFlE/SoMoSe #20: Arduino example still up to date for version 3.4?

R4M5E5, 2024-06-29:
Hey :)
I'm currently struggling a bit to get my new SoMoSe working with my Arduino Pro Mini. For me the Serial output always says `averaged humidity = 0`.
Is the example provided in the repo still up to date for version 3.4?
https://github.com/BeFlE/SoMoSe/blob/fcd1b907ee02416727ecbd59eb8bc70b98bcb69b/Arduino/Demo.ino#L1-L42

R4M5E5, 2024-06-30:
So far I found out that my sensor is in low power mode per default, according to the manual I need to call the `startMeasurement` function and wait 250ms and then check the `isMeasurementFinished` function. I still have the problem though that most of the time the measured average humidity is 0. Sometimes I got a value of 1 or 2, even when the sensor is in relatively wet soil. The value is a byte so theoretically values up to 255 could be possible.
Hardware version is 3.40 and firmware version is 1.10
Any idea what might be the problem here? The temperature value that I get from the sensor seems to make sense. It changes if I take the sensor in my hand and it usually is the same value as the temperature in the room.

BeFlE, 2024-06-30:
Hi, 

Can you check whether the reference values are set as specified in the data sheet?
I have noticed that there seem to be problems sometimes when initializing the sensor for the first time.
Sometimes the wet reference value is initialized to 0xFFFF. This should be corrected. This would also explain why the low power mode is active 
Tomorrow I can give you another code for this, unfortunately I'm not near my PC today.

Best regards 
Christian

R4M5E5, 2024-06-30:
Thank you, this seems to be the issue. The `SoMoSe_getReferenceWet` returns -1 which is 0xFFFF I guess. I will try setting it to 8000 as specified in the manual :)

BeFlE, 2024-07-01:
Hi,

yes the reference values have changed in the new version.
When first plugged into the ground, the sensor value will rise for a few hours. After that, however, it reacts quickly to changes. So did you read it once after a longer period of time in the soil? If the value is still too far off, you can also calibrate the reference value for your purposes.

Best regards
Christian

R4M5E5, 2024-07-01:
Is there a documentation of this specific behavior that the reference value is automatically determined when first plugged into the ground? From your last post it sounds to me that it works somehow like this:
When first plugged into the ground the sensor will automatically calibrate it self and update the reference values. After the calibration phase the reference value is stable.

How do I reset this calibration state? When is the calibration phase finished? Is the calibration also done in the low power mode?
I'm asking because I'm building a battery powered device.

Up until a couple of days I was working with the old SoMoSe v.2.2 where my goal was to let my microcontroller(arduino pro mini) be in deepsleep most of the time and periodically wake up. I wanted to only supply the SoMoSe with power while my microcontroller is awake, then take a measurement and turn off the power of the SoMoSe and let the microcontroller go back to deepsleep. I was planning on using as Mosfet/Transistor to do this. 
Now with the knowledge that there is this calibration phase and that we now have a sleepmode I'm not sure if this is still the best approach. Can you give me more information how the sleepmode is supposed to be used?