BeFlE/SoMoSe #21: Sensor displays strange values

Tuxtom007, 2024-07-01:
Hello,

I have experimented with two sensors (V2.2) on an ESP32 under ESPHome and am surprised at the values displayed.
The dry sensor shows 1% - ok, but if I put the sensor in a glass of water, it only shows 45 to 51%, shouldn't that go up to around 100%?

If so, how can you correct the values via the ESPHome YAML if necessary - I use the examples from the code example here from Github.

Thanks in advance

BeFlE, 2024-07-01:
Hi,

I'm not familiar with ESPHome and unfortunately I can't help at this point.
But you can use the reference values to calibrate the sensor more precisely if you wish. 
As an option, you can use my [SoMoSe firmware](https://github.com/BeFlE/SoMoSe/tree/main/SoMoSe-MQTT-ESP32_Firmware) for ESP32, adjust the values and then switch back to ESPHome, as the reference values are stored persistently. 
You just have to make sure to reset the I2C address to default, as my firmware changes it automatically.

Best regards
Christian

Translated with DeepL.com (free version)

Tuxtom007, 2024-07-01:
Hello,

thanks, I'll give it a try.
It's still a test installation at the moment anyway, so I can test it.

I had only used the sample code from the ESPHome folder and adapted it to my configuration.

best reagards
Thomas