BeFlE/SoMoSe #22: SoMoSe sensors stopped working after 1 day of operation, values are all over the place

Virtualizer, 2024-07-05:
Initially, I used the ESPhome firmware and integrated 4 SoMoSe sensors on one ESP32 by changing the I2C addresses.
For about 1 day all has been working fine, then I realized that the values are all over the place.
After that I installed the MQTT firmware on another ESP32 device and configured the sensors on that.
At this point I already got the result, that the sensors would not always be displayed in the sensors tab.
After configuration for the 4 sensors have been completed I opened the Home tab and got more or less the same results as with the ESPhome firmware. The sensors are flip-flopping online/offline and some strange values are reported.

See attached video:
https://github.com/BeFlE/SoMoSe/assets/175888/89473d21-effe-4a9b-b5c7-3aecb72981ca

BeFlE, 2024-07-08:
Hi,

There seems to be a problem with the hardware configuration.
Are the sensors supplied with 3.3V?
Are pull-ups being used? Are they connected to 3.3V?
If none are used, this should be taken into consideration.

Best regards
Christian

Virtualizer, 2024-07-08:
Ok, I will check with external pull-ups.
Do I have to use non-standard GPIO pins then, so that the internal pull-ups are not activated?

BeFlE, 2024-07-08:
You can use the same pins. If the internal pull-ups are activated, in the worst case you have the external ones in parallel. This is why they should not be smaller than 4k7.

Virtualizer, 2024-07-09:
Just installed the external pull-ups and the sensors are working now, thanks.
The readings of one of the 4 sensors are way too low regarding the others and the actual humidity of the soil.
About 1500 vs 4400.

poker-spt, 2024-07-30:
@Virtualizer Could you please upload a picture with the pull-ups installed?

Virtualizer, 2024-07-30:
@poker-spt 
I installed following this scheme:
[image]

closed 2024-08-10 by Virtualizer