BeFlE/SoMoSe #23: Does the supply voltage have an effect on the measured moisture value?

R4M5E5, 2024-07-13:
According to the datasheet the supply voltage can be anything between 2.5V to 5V. I'm planning to create a device which is supplied from alkaline batteries which have a severe voltage drop over time. Let's say I use 3 alkaline batteries then I will start at 3 \*1.5V=4.5V and when the batteries will be empty the voltage will be around 3*0.9V=2.7V
So the voltage will decrease from 4.5V to 2.7V over time. So my question is will the same moisture sensor give me the same moisture values for the same soil when supplied with 4.5V vs when supplied with 2.7V? Or in other words, does the supply voltage (which can be between 2.5V and 5V) have an effect on the measured moisture value?

Cheers,
Olli

BeFlE, 2024-07-14:
Hi,

Yes, the voltage has an effect on the measured moisture.
However, the SoMoSe v3.4 measures its operating voltage and compensates this effect to a minimum.
This means that the SoMoSe v3.4 can be used well for your application.

R4M5E5, 2024-07-14:
Awesome, thank you for the quick reply :)

Cheers,
Olli