BeFlE/SoMoSe #25: Questionable Moisture Values

tps42, 2024-08-01:
I am currently experimenting witch 2 SoMoSe Sensors (v2.2). I callibrated them (put in glass of water = 100%, in air=0%). When i put them in a very wet plant pot i see strange values. Sometimes i get a reading of about 25% - sometimes i get a reading of about 10% - none of them seem to correllate strongly to the real moisture as i can water the plant w/o noticeable change in readings. Sometimes i can even see a dropdown of the values while i feed in the sensor in the soaking wet pot. 

Any ideas? Electromagnetic interference? (LED light installed) "Wrong" soil? 

Thanx in advance!

riegera2412, 2024-08-30:
It is more likely the calibration. While many users calibrate using dry air and pure water this modus operandi for calibration is just wrong!

It would be better to calibrate the sensor in dry and saturated soil. This way the dielectric properties of the soil will be taken into account during calibration. Also, soil in one pot might be different from another. There are different types of soil recommended for different plants, so it is best to calibrate the sensor for each plant individually.

André

tps42, 2024-08-31:
sounds reasonable as you say it.... only the factory default triggered me to use the water-air-method. the defaults will be far far away when callibrating with wet and dry soil. using air and water it was only "a bit off" - i did not believe the factory defauts were so senseless...-Diese Mail wurde von unterwegs unter Anwendung aktueller Technologie gesendet. Daher ist inhaltlich mit Rechtschreib- wie Syntaxfehlern, ungewöhnlichem wie sinnlosem Satzbau und Aussageverfälschungen oder -invertierungen durch intelligente Spracherkennung, kontextsensitive Recjtschreibkorrwktue oder menschlicher Daumengröße zu rechnen. Diese Übertragungsfehler sind systemimmanent und nicht dem Autor zuzuschreiben.

"target": "https://github.com/BeFlE/SoMoSe/issues/25#issuecomment-2320329611",
"url": "https://github.com/BeFlE/SoMoSe/issues/25#issuecomment-2320329611",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

tps42, 2024-09-01:
the maximum raw value i can measure in soaking wet soil is about 800 with a variance of +- 200... this seems to be relatively small value compared to defaults and not very precise... what max raw values du you get in wet soil?-Diese Mail wurde von unterwegs unter Anwendung aktueller Technologie gesendet. Daher ist inhaltlich mit Rechtschreib- wie Syntaxfehlern, ungewöhnlichem wie sinnlosem Satzbau und Aussageverfälschungen oder -invertierungen durch intelligente Spracherkennung, kontextsensitive Recjtschreibkorrwktue oder menschlicher Daumengröße zu rechnen. Diese Übertragungsfehler sind systemimmanent und nicht dem Autor zuzuschreiben.

tps42, 2024-09-08:
Ok... got my issue tracked down (callibrating in wet and dry soil did not solve the problem):
It seems the sensors simply dont work at all in my environment!
I use a dimmable LED (150W) in 0,7m distance from the sensor. Whenever the light turns on moisture values go up by about 25%... 
You can see values of a week in attached image (blue moisture, red temperature, light turning on at 7 am an off on 11 pm - watering on Monday and Thursday) - values are totally senseless.

any ideas?
[image]