BeFlE/SoMoSe #27: Standby mode vs one time measurement

R4M5E5, 2024-09-19:
Hi,

so I understand that since version 3.4 we have a standbymode with a current of 16u. I would like to understand why such a standby mode is necessary and we just can't turn off the sensor completely and just turn it on for a measurement?
It would like to build a device which is powered by a CR2032 battery like the HHCC Soil Moisture Meter (Mi Flora Sensor), for that a standby current of 16u is already quite high considering also that the actual measurement, microcontroller and rf chip will also consume additional power.

Best regards

BeFlE, 2024-09-20:
Hi,

Of course you can also simply switch the sensor off and then on again when required. 
This also works well and is probably even best for your application.
I mainly introduced the energy-saving mode in order to be prepared for a variant with a solar cell. The 16uA are not that critical there. 

Best regards
Christian

R4M5E5, 2024-09-21:
Hey,

just out of curiosity, what are the benefits of keeping the sensor in standby vs turning it off completely between measurements? I mean we are "loosing energy in the form of a 16ua current" so there is probably an advantage of this method which justifies this energy consumption. I'm just asking to have all the details here to pick the right approach for me and I'm a little bit worried that I'm overlooking something.

Best regards
Olli

BeFlE, 2024-09-24:
Hi,

the advantage of keeping the sensor in standby is that it reacts faster and the measured value is a little more accurate and reproducible. 
In addition, undesirable behavior can occur when switching off, e.g. if several sensors are connected to the bus and consume current, the switch (transistor, MOSFET or similar) has a resistance and an undefined voltage drop or ripple may occur.

best regards
Christian