BeFlE/SoMoSe #29: ESPHome deprecated Custom Components. Please Create External Component.

hakunamatataPino, 2025-02-06:
Sadly, ESPHome deprecated Custom Components. It is required to migrate to External Component.
Sources:
[ESPHome Custom Sensor](https://esphome.io/components/sensor/custom.html)
[ESPHome External Component](https://esphome.io/components/external_components)

BeFlE, 2025-02-07:
I will be studying the topic in the near future and trying to integrate the somose.

joe16joe16, 2025-04-25:
@hakunamatataPino check the merged PR with a minimal version, hopefully @BeFlE are so kind soon and add the other features beside temperature and basic moisture read out like:

- set I2C address to use more than one sensor at the same I2C bus
- set wet
- set dry

I also checked his MQTT firmware, which looks great, but in my case I just need the sensor on a S3 Pro which already controlling my four zone backyard sprinkler system with esphome.

My "dream" are a independent esphome controller which is able to water by a internal time scheduler and just enough for the plants and current environment, this needs beside air temperature, pressure (BME280) and brightness (BH1750) a reliable moisture sensor like this!

BeFlE, 2025-05-06:
I will start integrating the functions in the next few days.

But first I have to get familiar with ESPhome and set up a test environment.

Does anyone have any tips on how best to set up a test environment? preferably with VSCode

BeFlE, 2025-05-13:
Hi,
short update:
I've been trying to build a stable development environment for a few days now.
However, there are still a few problems.
I think I've managed to track down the problems and am waiting for a new Raspi, as my old one is too weak. It should arrive tomorrow and hopefully I can start developing once it's set up.
I have created a test repo [here](https://github.com/BeFlE/testing), where I will always upload the current status if anyone is interested.

Best regards Christian

BeFlE, 2025-05-22:
Hi, 
@joe16joe16, @hakunamatataPino I have now published the code for the external component. During my tests I could not find any major errors, if you notice anything else please let me know.

best regards 
Christian

closed 2025-05-28 by BeFlE